ut-robotics/picr21-team-kurgimopeed #16: [Mechanics] - Omniwheel roller extrusions should not cover the whole wheel

Akustav, 2021-10-29:
The extruded part locating the rollers should not touch anything but the internal part of the bearing. You would need to make a cut (red) and everything after that should be flat and further away from the rollers (blue)

[image]

Otherwise you can also use a conical extrusion similar to this:
[image]

Akustav, 2021-11-06:
Reopening. I would also remove these parts. They can touch the moving part of the bearing. 
[image]

MattiasAvi, 2021-11-07:
Fixed This issue

closed 2021-11-07 by MattiasAvi