ut-robotics/picr21-team-kurgimopeed #33: Separate code more

ReikoR, 2021-12-11:
Right now `ImageProcess` class is inheriting `RSCamera` class and internally using `DrivingLogic` class instance. It would be better to clearly separate different classes so that they can function independently.

Generic example on how to keep different classes separate:
```python
class RobotController:
    def run(self):
        while not self.is_stopped:
            rgb_image = self.camera.get_rgb_image()
            objects = self.image_processor.find_objects(rgb_image) # returns balls, baskets
            motor_speeds = self.driving_logic.run_logic(objects)
            self.motor_driver.set_speeds(motor_speeds)
```

Above code is only partially based on existing code and omits a lot of functionality, but should give a general idea.

joshmmst, 2021-12-16:
hello
, we did some big changes in f58cd061d8bf84e7fb61010434024d1e80bd4f52. Everything should be O.K! Also added referee client.

joshmmst, 2021-12-16:
we kept motor driver on separate core with multiprocessing and referee server creates an event loop thread but rest is one thread pipeline. please do check RobotController class

ReikoR, 2021-12-17:
https://github.com/ut-robotics/picr21-team-kurgimopeed/blob/8249a71791fdd2d88afe97c632eb1babe7b56f4a/backend/src/RobotController.py#L25-L28

https://github.com/ut-robotics/picr21-team-kurgimopeed/blob/8249a71791fdd2d88afe97c632eb1babe7b56f4a/backend/src/DrivingLogic.py#L42

You are also instantiating `RefereeClient` in `DrivingLogic`, but it's not used there.

jakoai, 2021-12-18:
Removen RefereeClient from DrivingLogic

closed 2021-12-18 by ReikoR