ut-robotics/picr21-team-kurgimopeed #35: [Mechanics] - thrower angle adjust could use previous iteration

Akustav, 2021-12-13:
I would change the thrower angle-adjust a little. I would make it so that the movement of the internal ramp is guided by a separate arm that extends from the center of the ramp (also the center of the motor), like this:

[image]

Such a system allows for a very smoothly moving internal ramp. Do you have a motor with a shaft?

Additionally the simplest angle adjust would be something like this:
[image]

Where the blue piece has a slot in it that goes into a pin that is attached to the moving ramp. If you move the servo up-down then the pin slides in this slot and allows the ramp to move either up or down.

Akustav, 2021-12-13:
You can also look at how the servo arm was used in conjunction with the blue piece.

Akustav, 2021-12-14:
Additionally it is unclear how stuff attaches to the servo since the servo horn is not in the model.

[image]

Akustav, 2021-12-21:
I would also beef up the arm. It should bend as little as possible while throwing:

[image]

MattiasAvi, 2021-12-21:
Made this part bulkier.

Akustav, 2021-12-21:
How do you plan to assemble it? I would make it so that you fasten the arm to the servo horn and then screw this assembly to the servo. Currently you need to do the final assembly with the servo attached that can make it more difficult.

I would have a central hole in the arm that lets you attach and remove the assembly simply.

MattiasAvi, 2021-12-21:
Thanks for this suggestion. It would have been quite annoying to assemble it. Made a hole in the center.

closed 2021-12-22 by Akustav