ut-robotics/picr21-team-kurgimopeed #37: [Mechanics] - how are internal ramp components attached?

Akustav, 2021-12-13:
How are these pieces attached to eachother?

[image]

MattiasAvi, 2021-12-14:
Stationary parts are attached with bolts and spacers. Do we have any long rods at least 65 mm of length in Delta? Preferably around 3 mm diameter.

Akustav, 2021-12-14:
I dont think we have such a rod. Why not use spacers instead? Would make assembly simpler and more robust. Additionally, why have the slot instead of 2 holes?

[image]

MattiasAvi, 2021-12-21:
I am going to use hex standoffs for spacing. Most spaces are multiples of 5.

Akustav, 2021-12-21:
OK

closed 2021-12-21 by Akustav