ut-robotics/picr21-team-kurgimopeed #38: [Mechanics] - button and power lead input out of spec

Akustav, 2021-12-13:
The power input for the computer and the switch ports are out of spec:

[image]

Their requirements can be found here: https://digilabor.ut.ee/index.php/Basketball_robot_mechanics_design#PC_power_adapter_socket

Additionally, your camera mount blocks the hole to the power socket.

MattiasAvi, 2021-12-14:
Switch and button holes should fit specifications. Left some room at shorter ends of switch to fit it in better.

Akustav, 2021-12-14:
You left space here? The Length is 32 mm
[image]

closed 2021-12-21 by Akustav