ut-robotics/picr21-team-kurgimopeed #43: [Mechanics] - roller design questions

Akustav, 2021-12-14:
Why is the roller asymmetrical? Based on my experience it has no big benefit compared to a normal round one since the ball has a rubbery texture and will stick to most surfaces.

I would suggest something similar to this:

[image]
[image]

This has the advantage of being simple to attach and remove, the servo horn should not be too much in the way. Additionally I would suggest making the whole roller girthier, to increase its strength. You might ram things with it.

MattiasAvi, 2021-12-21:
I found asymmetrical design from one of the thrower design examples. But your idea seems much better and I made something similar.

Akustav, 2021-12-21:
It is true that I have experimented with the asymmetrical design. It did not bring any tangible benefits compared to a round and symmetrical design.

closed 2021-12-21 by Akustav