ut-robotics/picr21-team-kurgimopeed #46: [Mechanics] - electronics board interference with motor encoder caps

Akustav, 2021-12-21:
The encoder caps and the electronics board intersect:

[image]

MattiasAvi, 2021-12-21:
Draw more precise PCB shape and now there\s no interference.

Akustav, 2021-12-21:
I think you can not place the board on the bottom flat since it has headers that go through the board. Also if it lifts this might be an issue:
[image]

MattiasAvi, 2021-12-21:
This should not be an issue because our robot's encoders are stripped from the case and in reality are quite smaller.

Akustav, 2021-12-22:
If you say so

closed 2021-12-22 by Akustav