ut-robotics/picr21-team-kurgimopeed #48: [Mechanics] - weird order of operations

Akustav, 2021-12-28:
Why is the last operation a slot? It might rip the piece off the substrate since it is held on just by tabs:

[image]

Akustav, 2021-12-28:
Similar issue is with the 2DContour2, where too much is in one operation. It might mille the OD out first, then does the internal slot, that might break off:

[image]

MattiasAvi, 2021-12-28:
> Why is the last operation a slot? It might rip the piece off the substrate since it is held on just by tabs:
> 
> [image]

My bad. Forgot to move it up. Managed to merge this contour with another and use 3 mm mill.

MattiasAvi, 2021-12-28:
> Similar issue is with the 2DContour2, where too much is in one operation. It might mille the OD out first, then does the internal slot, that might break off:
> 
> [image]

By changing some parameters I was able to get rid of the center slot. Now there are no tabs needed.

closed 2021-12-29 by Akustav